dhalperi/linux-80211n-csitool-supplementary #419: Will it not work on ubuntu 16.04 ?

priyankar2017, 2019-03-26:


shawon007bd, 2019-03-26:
NO. It only works with "Linux kernel version between 3.2 (e.g. Ubuntu 12.04) and 4.2 (e.g. Ubuntu 14.04.4)". Please check 1st two lines from this [Installation Instructions](http://dhalperi.github.io/linux-80211n-csitool/installation.html)
We tried it with ubuntu 14.04.6 and it didn't work.

kallami, 2019-04-02:
@priyankar2017 were you able to get the tool to work with ubuntu 16.04?

jadhavhninad, 2019-06-14:
We successful got it working on Ubuntu 16.04 (kernel 4.15) as part of our project and are working on pushing the updated code.

priyankar2017, 2019-06-14:
Hi Ninad Jadhav,

Could you please provide the details of the hardware (Laptop name along
with the model number, Wifi Access Point Name along with model number)

ExcelYeh, 2019-11-24:
kernel is the only limitation. Some do success on Ubuntu18. However some commands on 18.04 is different from installation.index given. Therefore, 16.04 is the highest choice if you don’t know much about the commands in Linux.