dhalperi/linux-80211n-csitool-supplementary #420: How to plot CSI amplitude vs Time

abhinav2938, 2019-03-28:
Hi,
I am able to receive the CSI amplitude vs no. of packets graph. But I don't know how to plot it in time domain. A similar issue has been covered(https://github.com/dhalperi/linux-80211n-csitool-supplementary/issues/189) previously but there also the end result was that the graph of CSI amplitude vs no. of packets gives data in time domain, but I want it specifically to plot amplitude vs time because I need to further process it for generating spectrogram. I am attaching one screenshot of my plot.
Any help is highly appreciated.
[image]

Bowloforanges, 2020-03-05:
if you send a determined number of packets per second then you know how many packets are equal to one second in that graph, maybe you could try that approach!

susan381, 2020-05-15:
@abhinav2938 
Hi, could you please tell me how you plot the CSI amplitude vs. no. of packes?
I've asked the same question here: https://github.com/dhalperi/linux-80211n-csitool-supplementary/issues/189

Many thanks!

Bowloforanges, 2020-05-15:
you'd need to write a MATLAB script that allows you to take the data for each packet in the CSI file and graph it, alternatively If you provide me with an email I'd be glad to share mine with you.

susan381, 2020-05-15:
@Bowloforanges  Thanks for offering help. Appreciate it.
Here is my email address: susnna15@gmail.com

Looking forward to hearing from you.
Thanks!

susan381, 2020-05-16:
@Bowloforanges  Haven't got any email from you yet. Please let me know when you sent it out. Thanks!

Bowloforanges, 2020-05-17:
I've sent you the email, sorry for the delay.