dhalperi/linux-80211n-csitool-supplementary #441: Does anybody knows how to bind two NICs in one receiver?

zhangshaohu, 2019-10-15:
I have one laptop with two wifi adapters. I  want to collect csi data simultaneously using two NICs.
The connection looks like the paper Phaser did. Does anybody know how to run command binding two NICs in monitor mode?

purohitank, 2019-10-26:
I also have the same question.

purohitank, 2019-10-29:
Dear Zhang, 

I came across this project which should prove useful. https://gitlab.com/wifisensing

The group has implemented reception of CSI from multiple NICs.

I have not tried the tool myself. Let me know how it turns up.

Regards,
Ankit

zhangshaohu, 2019-10-31:
I don't have any idea how to use this tool PocoSense after reading their paper and codes. Not well-maintained codes and tutorial.